GameboyZero/GabozePocaio #3: Lakka.tv on 2.4" ILI9341 18 Pin Display

32teeth, 2018-05-07:
[image]

# Introduction
> Want some free? Take a look at the bounties, get in touch

## [Lakka.tv](http://www.lakka.tv/)
> Lakka is a lightweight Linux distribution that transforms a small computer into a full blown retrogaming console.

### Description
Currently, the Gaboze Pocaio uses *fbtft_device* module to run on the 2.4" TFT LCD with ILI9341 driver. This is done by installing and running the [Frambuffer Copy](https://github.com/tasanakorn/rpi-fbcp) from [tasanakorn](https://github.com/tasanakorn/)

### Challenge
We love all Open Source OS platforms for gaming. Lakka.tv currently has no built in support for 18 pin SPI TFT LCD Interface. We could use your help!

#### TFT Pin Mapping
| PIN | Function | Required |
| ---- | ---- | ---- |
| 1 | GND | YES |
| 2 | RESET | YES |
| 3 | SCL | YES |
| 4 | D/C | YES |
| 5 | CS | YES |
| 6 | SDA | YES |
| 7 | SD0 | YES |
| 8 | GND | YES |
| 9 | VCC | YES |
| 10 | LEDA | YES |
| 11 | LEDK1 | YES |
| 12 | LEDK2| YES |
| 13 | LEDK3 | YES |
| 14 | LEDK4 | YES |
| 15 | XL | NO |
| 16 | YU | NO |
| 17 | XR | NO |
| 18 | YD | NO |

#### TFT Device Reference
[image]

### Reward
> As a reward other than fame an notoriety within the Game Boy Zero community, we will:

* Send you a fully built Gaboze Pocaio (at our cost)
  * Assembled Production PCB
  * Raspberry Pi Zero W
  * LiPo Battery
  * SD Card
  * Shell and Buttons
* YouTube Video showcasing your awesome work
* Provide a distrution to the community of your ISO

juj, 2018-06-18:
Tested out fbcp-ili9341 on Pi Zero with ILI9341 via SPI, and it does work nicely:

[image]

one tricky part is that Lakka is based on LibreELEC, which is a crippled Linux distro that does not come with a package manager that would be able to install git, CMake, or a C++ compiler to it, so fbcp-ili9341 must first be built on e.g. a Raspbian Stretch based Pi 3B/Zero system, and then copied over to the Lakka SD card. I did so with

```bash
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
mkdir build
cd build
cmake -DPI_ZERO=ON -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=25 -DSPI_BUS_CLOCK_DIVISOR=6 ..
make -j
```

and then copied resulting `fbcp-ili9341` executable file over to the Lakka SD card in directory `fbcp-ili9341/`, and enabled SSH on it in the launcher, logged in as `root`/`root`, and

```bash
cd fbcp-ili9341
chmod +x fbcp-ili9341
./fbcp-ili9341
```

To get fbcp-ili9341 booting up at startup, adding a line `/storage/fbcp-ili9341/fbcp-ili9341 &` in file `/storage/.config/autostart.sh` did the trick. (https://wiki.libreelec.tv/autostart.sh)

32teeth, 2018-06-18:
@juj thank you for the Lakka instructions

i have attempted a quick implementation of yours following instructions with no success (white screen)
are you adding anything to options in the line
```
cmake -DPI_ZERO=ON -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=25 -DSPI_BUS_CLOCK_DIVISOR=6 ..
```
or are the two dots intentional or does it mean 'more options'

thanks

juj, 2018-06-18:
The `..` denote "up one directory", i.e. to locate the path where fbcp-ili9341 source directory CMakeLists.txt file resides.

When getting a white screen, the checklist at https://github.com/juj/fbcp-ili9341/#i-am-running-fbcp-ili9341-on-a-display-that-was-listed-above-but-the-display-stays-white-after-startup is good to verify. In particular, the pin assignments via the `-DGPIO_TFT_DATA_CONTROL=` and also `-DGPIO_TFT_RESET_PIN=` if there's one used on the display.

32teeth, 2018-06-18:
thanks for the clarification.
will test later this evening

juj, 2018-06-19:
Great to hear that it works!

> @juj did you still change your config.txt

I did not change HDMI resolution for above testing - running `tvservice -s` in console gave

```
Lakka (official): 2.1 (RPi.arm)
Lakka:~ # tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
```

when I did not have a HDMI display connected. When a display was connected, it was 1920x1080. That does work, although for best performance if simultaneous HDMI out is not needed, configuring 320x240 is very much preferred, since the workload on the GPU will then be much smaller.

> Is there a way to supress the statistics from the compiled version?

Statistics can be removed by commenting out `#define STATISTICS` in `config.h`. I now pushed a change to the repository that allows also passing `-DSTATISTICS=0` to CMake to disable statistics. At some point I am looking to migrate to more run-time like options for dynamic toggling of statistics, but for now compile time settings have been the simplest.

Btw, I noticed in https://sudomod.com/forum/viewtopic.php?t=5792#p60499 a comment about display possibly being off-centered. If this is still a thing in final production units that the screen size can not perfectly match the enclosure, then fbcp-ili9341 does have a method to "cut off" rows/columns of pixels from the display active visible area, making effectively dead area on the edges. You can activate that by changing the numbers `0` in https://github.com/juj/fbcp-ili9341/blob/master/display.h#L25 to something > 0. For example `#define DISPLAY_COVERED_LEFT_SIDE 10` would cut off 10 pixels on the left (or actually from bottom, if pretending the display is installed in portrait orientation).

That option can then be paired with `-DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON` or `-DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON` options to adjust how the scaling will then be performed (crop, scale with aspect, scale without aspect).

closed 2018-06-28 by 32teeth