GameboyZero/GabozePocaio #9: Missing Eagle Library?

Uplinkpro, 2018-07-03:
Every time I try to open this up in Eagle I'm missing a "GBA" library, I've added all the lbrs you have provided with no luck.

32teeth, 2018-07-03:
Hmm, the only one part that uses that is the volume potentiometer. I’ll look into it

32teeth, 2018-07-03:
Ah, you are opening the layout file. It has been stripped of most components and is indented for “layout” purposes.

Uplinkpro, 2018-07-03:
Gotcha , the problem with the board now is if you hit the up button it will also hit the right or left button if you make that little change you see I made in the layout, that should correct the issue I believe, Also would be neat to try to use a USB Type-C connector on this but that's another thing all together.

32teeth, 2018-07-12:
| [image] |
|:--:|
| *[Conductive Button Library](https://github.com/GameboyZero/GabozePocaio/blob/master/lbr/ConductiveButtons.lbr)* |

closed 2018-07-12 by 32teeth

lipomos, 2018-07-14:
hi 32
Your great job is very good
so I want the perfect full eagle file.
Can you do that? please

goc335, 2018-08-02:
The GBA library is still not available on the library folder. And many og the components are missing og the schematic in the eagle files.